maxgerhardt/pio-esp32-esp8266-filesystem-downloader #2: Feature request: littlefs for ESP32

Jason2866, 2021-01-13:
Great addon! It would be great if you add littlefs for ESP32. We use it for ESP32 already. See [Tasmota](https://github.com/arendst/Tasmota)

wimmatthijs, 2021-04-16:
Also kindly requesting this feature :)

3gyptian, 2022-06-01:
ditto!

Munger, 2023-07-23:
Has there been any update on this feature?

Endorfin35, 2023-10-04:
great... any plans for esp32 and littlefs?

Jason2866, 2023-10-04:
In the meantime we did a workaround. To unpack files correctly mklittlefs needs to be compiled with a different setting. The updated mklittlefs is in pio registry https://registry.platformio.org/tools/tasmota/tool-mklittlefs
and the modified script von Max is [here](https://github.com/arendst/Tasmota/blob/development/pio-tools/custom_target.py)

maxgerhardt, 2023-10-05:
Yeah I'll try and update this repo's scripts today so that all current possible filesystems (SPIFFS, LittleFS, FatFS) are supported properly. Shouldn't be that much of a deal.

Jason2866, 2023-10-16:
@maxgerhardt We updated our script (based on yours) to read the LittleFS addresses from the device itself (Tasmota can change partitions on the fly) to make sure the are always correct. No more partition.csv is needed/used.
https://github.com/arendst/Tasmota/blob/development/pio-tools/custom_target.py