maxgerhardt/pio-esp32-esp8266-filesystem-downloader #4: Bad powershell command for extraction of downloaded binary

mark-hahn, 2022-06-18:
I am running a platformio task in vscode.  It runs `download_fs.py`.  The script tries to run a mklittlefs command but powershell barfs ...

```
> "C:\Users\mark\.platformio\packages\tool-mklittlefs\mklittlefs" -b 8192 -p 256 --unpack "unpacked_fs" "C:\Users\mark\apps\tiny\downloaded_fs_0x300000_0xfa000.bin"
At line:1 char:65
+ ... sers\mark\.platformio\packages\tool-mklittlefs\mklittlefs" 
-b 8192 -p ...
+
~~
Unexpected token '-b' in expression or statement.
```
The empty folder `unpacked_fs` was created and the downloaded binary `downloaded_fs_0x300000_0xfa000.bin` was in the root of the project folder. 

I don't know anything about powershell and I can't figure out what is wrong with that command.  I googled a lot.

Everything worked in the past before I reinstalled windows 10, vscode, and platformio and ran the project I cloned that used this.  Can someone help me?

maxgerhardt, 2022-06-18:
Likely it doesn't like that the binary is quoted or there is the `.exe` missing. I'll check. I tested this with the default terminal set to `cmd.exe`.

mark-hahn, 2022-06-18:
> <maxgerhardt/pio-esp32-esp8266-filesystem-downloader/issues/4/1159387169@
> github.com>

I tried changing the quotes on everything but I might have missed the right
combination.  I'll do more experimenting.

mark-hahn, 2022-06-19:
I give up.  We need a powershell (or cmd) expert.  It appears to be a simple syntax error.  (I hate windows).

maxgerhardt, 2023-10-05:
This should have been fixed by #7, I will retest this in Powershell.