maxgerhardt/pio-esp32-esp8266-filesystem-downloader #12: getting .platformio/packages/tool-mklittlefs/mkspiffs__ no such file or directory after FS download

kevinlieb, 2024-05-13:
I'm on a Mac.  I integrated the .py script into my project but I get:

Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: c0:4e:30:12:ad:c0
Uploading stub...
Running stub...
Stub running...
1572864 (100 %)
1572864 (100 %)
Read 1572864 bytes at 0x00670000 in 176.1 seconds (71.5 kbit/s)...
Hard resetting via RTS pin...
Unpack files from filesystem image
/bin/sh: /Users/lieberk2/.platformio/packages/tool-mklittlefs/mkspiffs__: No such file or directory
Extracted 0 file(s) from filesystem.

Did I miss a step?
Is this a Mac specific problem?

kevinlieb, 2024-05-13:
Should clarify I put this in platformio.ini

extra_scripts = download_fs.py
custom_unpack_dir = /Users/lieberk2/littlefs

I'm using ESP32-S3 on an M5Stack Dial

JacoAnnandale, 2024-06-27:
Getting more or less the same problem:

_C:\Users\jacoa\.platformio\packages\tool-mklittlefs\mkspiffs_${PIOPLATFORM}_${PIOFRAMEWORK}"_' is not recognized as an internal or external command,
operable program or batch file

I am using Win10 64 bit

mcsarge, 2024-08-12:
Same for me:

'"C:\Users\msargent\.platformio\packages\tool-mklittlefs\mkspiffs_${PIOPLATFORM}_${PIOFRAMEWORK}"' is not recognized as an internal or external command,
operable program or batch file.

DavidStacer, 2024-10-18:
Same error message for me on Windows.  Looking through the scrips, where are the PIOPLATFORM and PIOFRAMEWORK variables coming from?

I put in some print statements and these two variables are not defined.